GrapheneOS/Camera #464: Camera crashing since version 69

Aquathing, 2024-08-09:
My camera has been crashing since version 69 (or 68, cant remember properly). 
It happens when:
- I open the camera app
- I use the camera app and zoom
- I open the camera from another app such as Telegram

The crashes are occasional and I can't find what triggers them, but I've also noticed that when taking pictures from Telegram the camera can take some seconds to actually shoot the picture.

[android.hardware.camera.provider@2.7-service-google crash 873980f97da6.txt](https://github.com/user-attachments/files/16563808/android.hardware.camera.provider%402.7-service-google.crash.873980f97da6.txt)
[android.hardware.camera.provider@2.7-service-google crash 769514d78669.txt](https://github.com/user-attachments/files/16563809/android.hardware.camera.provider%402.7-service-google.crash.769514d78669.txt)

MHShetty, 2024-08-09:
Hi @Aquathing,

I'm failing to reproduce this on my end despite trying the given steps multiple times. The zoom just resets in my case while switching cameras that way with the Telegram app

Are you sure this is a regression from a specific version of the camera app and not any hardware/camera related OS component issue?

Aquathing, 2024-08-09:
> @Aquathing This is a hardware issue causing an OS crash, not a Camera app crash.

By "hardware issue" you mean this is caused by defective hardware (funny given how this happened just after it came back from RMA) or is it an hardware issue of every Pixel device?

Aquathing, 2024-08-10:
> Defective hardware. They likely gave you back a faulty one if it was after RMA. The replacements are normally different refurbished devices and they could have issues which got through their QA testing.

Sad to hear, but thanks Daniel 🤗

Aquathing, 2024-10-22:
Update: This morning my phone updated to 2024102100.

I have opened the Camera app something like 100 times, and could not reproduce the issue any more. Only thing that I've noticed is that (only once, for now) both the Google Camera and the GrapheneOS camera apps would not start. 

The GrapheneOS app manifested this issue by staying stuck on the splash screen.
The Google Camera app manifested this by only showing the three UI buttons (gallery, but no preview, shutter button, and switch camera). Anything else was not shown.

The camera was still accessible through some other apps like Telegram, so this is probably not an hardware issue. A reboot fixed this.

Aquathing, 2024-10-24:
It started happening again, I guess it's RMA time. Sorry.