GrapheneOS/Camera #481: Image Processing Takes Forever

garcia1987, 2024-11-13:
Pixel 9 Pro

This issue seems to happen on both the Graphene Camera and Pixel Camera.

I will take a picture and it will take forever to process where I can see it. The Gallery will display blank icons for everything, using the files to browse folders in the system will sometimes not show folders. I have seen this go on for up to half an hour at this point.

So far I've tried the following to resolve the issue. Not much but I'm not well versed in what else to do.

I will force close apps (camera/gallery) for the pictures to appear and/or the thumbnails to show up. This has caused me to lose a photo or two.

I've had to restart the phone a couple of times from the above not working.

I have no issues running other apps or doing other tasks during this time. This is specifically only affecting the Gallery, Graphene Camera, and Pixel Camera.

As an aside I did try OpenCamera which will allow me to see the pictures I take in the gallery within 10 or so seconds.

I'm really curious if it is something I am doing or if there is just some heavy work going on behind the scenes to process these images and it takes a bit.

MHShetty, 2024-11-14:
Hi @garcia1987,

Could you please test the following version of our camera app for this issue? 

https://drive.proton.me/urls/K1HMFMRG88#67z8HZrVaMkD

garcia1987, 2024-11-14:
It actually appears to be worse now. The camera is unable to take consecutive pictures in a timely manner. I'm not button mashing (using 10 second intervals) but it's taking upwards of a minute for me to be able to take another picture after the first one.

The gallery is still doing the same thing, but now it's freezing up which is new.

MHShetty, 2024-11-14:
Ah that's really weird. Wanted to test if the new gallery implementation resolves the image not getting displayed but this seems to be a different issue altogether. 

Since this issue is impacting multiple apps, it might make more sense to shift it to the OS issue tracker, although it is still uncertain what the actual issue is with your device/OS build

garcia1987, 2024-11-14:
For reference
Build Number: 2024110700

Is there anything I can do on my end to help?

I'm not a dev but I do work desktop support and have been fiddling around with things for the better part of 20 years so I can definitely follow some instructions lol.

warreth, 2025-11-20:
I still have this issue on my pixel 8 Pro running the newest android 16 version.