GrapheneOS/Camera #485: Night mode not working

doomlaur, 2024-11-22:
I have a Google Pixel 9 Pro XL with GrapheneOS 2024111800 (the newest version at the time of writing) and Camera version 74  (the newest version at the time of writing). I'm very happy overall, except that the night mode does not seem to work in the Camera app. Switching to the night mode shows me a beautiful preview image, but pressing the button to actually take the photo never actually saves it. In other words, taking the photo never finishes. I see the following error when trying to open the Gallery from the Camera app:

> Please wait for the image to get captured before trying to open the gallery.

Pressing the button for switching the camera causes the app to show the following error message:

> Unable to capture image, error code: 3

Pressing `Show details` in that window shows me the following call stack:

```
osVersion: google/komodo/komodo:15/AP3A.241105.008/2024111800:user/release-keys
package: app.grapheneos.camera:74

x.S: Camera is closed.
	at y.r.a(Unknown Source:9)
	at x.Q.x(Unknown Source:12)
	at x.q0.B(Unknown Source:0)
	at E.h.C(Unknown Source:197)
	at E.h.A(Unknown Source:14)
	at androidx.camera.lifecycle.d.d(Unknown Source:53)
	at d1.p.t(Unknown Source:179)
	at k1.l.onClick(Unknown Source:542)
	at android.view.View.performClick(View.java:8036)
	at d1.e.onTouch(Unknown Source:129)
	at android.view.View.performOnTouchCallback(View.java:16445)
	at android.view.View.dispatchTouchEvent(View.java:16412)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3116)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2789)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3144)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3122)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3122)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3122)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3122)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3122)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3122)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3122)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2803)
	at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:458)
	at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1981)
	at android.app.Activity.dispatchTouchEvent(Activity.java:4541)
	at e.t.dispatchTouchEvent(Unknown Source:2)
	at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:416)
	at android.view.View.dispatchPointerEvent(View.java:16737)
	at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:7947)
	at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:7710)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:7106)
	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:7163)
	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:7129)
	at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:7295)
	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:7137)
	at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:7352)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:7110)
	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:7163)
	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:7129)
	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:7137)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:7110)
	at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:10214)
	at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:10165)
	at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:10134)
	at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:10356)
	at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:295)
	at android.os.MessageQueue.nativePollOnce(Native Method)
	at android.os.MessageQueue.next(MessageQueue.java:346)
	at android.os.Looper.loopOnce(Looper.java:189)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8744)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
	at com.android.internal.os.ExecInit.main(ExecInit.java:50)
	at com.android.internal.os.RuntimeInit.nativeFinishInit(Native Method)
	at com.android.internal.os.RuntimeInit.main(RuntimeInit.java:369)
```

Thank you in advance for your help, and thank you for the great work on GrapheneOS and the Camera app! :smile:

MHShetty, 2024-11-22:
Hi @doomlaur,

Thanks a lot for filing this issue and your kind words and appreciation!

How long do you wait until night mode finishes taking the shot? Does the issue occur even after waiting for 10-15 seconds? (considering the worst case scenario)

doomlaur, 2024-11-22:
Hi @MHShetty and thank you for your quick answer! :slightly_smiling_face: The issue does occur even after waiting for some time. I just tried it again and waited for 5 minutes (confirmed by the stopwatch) and it still did not finish.

doomlaur, 2024-11-23:
Perfect, thank you so much! :smile:

mrxx0, 2024-12-13:
@doomlaur Hi, should be fixed with the new camera version 75.

doomlaur, 2024-12-14:
I can confirm that it works with version 75. Thank you! :smile:

melo0187, 2024-12-14:
I'm using the Pixel 8 and with version 75 night mode does not take any pictures anymore. 

Is there something I can do about it? 
Or at least provide logs here? 

Thanks in advance!

melo0187, 2024-12-18:
Can confirm that a simple reboot made the issue I was facing go away. Thanks!