GrapheneOS/Camera #491: Frame rate drops, jerky video when recording multiple videos with GOS Camera

cullumsmith, 2024-12-14:
I've used the default GrapheneOS Camera for about 4 years with no issues. Starting a few months ago, I started having having the following issue recording video with the GOS camera:

I can capture one video without any problems. But capturing subsequent videos results in multiple lags/jerks/frame skips. The issue is still present on the recently released GOS camera version 75 (if anything, its actually a bit worse in this version).

The issue does **not** happen using Google's Pixel Camera app from the Play store.

What _appears_ to be happening is that the GOS Camera is performing some kind of lengthy post-processing after I end the capture, and the CPU becomes too busy to take subsequent videos (unless I reboot or wait a long time (say, a few minutes) between video captures).

I can lessen the problem by switching to a lower resolution, but the problem never goes away entirely.

How I can reproduce:

Take multiple ~15 second videos, one after the other (record, stop, record, stop, etc). After the 3rd/4th attempt, the resulting video has numerous lags where the video freezes for a few seconds. After the 5th/6th attempt, video capture almost stops working entirely. The phone becomes quite hot.

This happens for me on my Pixel 7 and a brand new Pixel 8a.

I can lessen the problem by switching to a lower resolution (1080p) and disabling image stabilization, but the problem never goes away entirely.

drammelt, 2025-01-20:
I wondered why I kept getting lag spikes when recording videos, it actually affects the recordings as well. I get 2-3 seconds of lag, especially when moving the camera it kind of feels like the stabilization is adding resistance to the camera movement on screen, it will suddenly jump from the previous position to the new one but smoothly if that makes sense. Then the resulting video looks normal but has glitches at these points like a bunch of frames were dropped.

drammelt, 2025-01-21:
It seems to only happen when using zoom + recording multiple videos in quick succession as the original report describes. Just for sanity I checked using GCam and I do notice a stickiness when zoomed in but it happens less often, is less noticeable and there is no lag. I also noticed that the phone did not heat up at all as it did with GOS Camera, even after many short or long videos. I had set GCam to use GOS Gallery app as well and am using fake GPlay service APK so there is no AI/cloud shenanigans going on.

I have tried resetting GOS Camera to factory version and update again and so far I cannot reproduce, and it did not heat up like it did before reset. I will check under "normal" conditions later which is outside in the sun + cellular (usually low reception at the beach) and recording waves/surfing which I guess due to colour flatness may cause issues with EIS/focus.

drammelt, 2025-01-24:
Confirmed for me at least, resetting GOS Camera to V74, then updating again fixed the issue with dropped frames/stalling when recording multiple videos. Guessing this is mainly due to the app data being cleared. Also guessing this might be an issue for where the OS was updated from 14->15?

However the camera while zoomed on the sea at least has a real hard time focusing and is very resistant to movement with EIS on and gives the impression of lag/slowness still. GCam on the other hand has no issues.

MHShetty, 2025-05-16:
Hi @cullumsmith,

Could you please confirm if you face this issue in the latest release? We could file an issue to the CameraX team for the same if that still is the case

Thanks a lot for your time and support!

cullumsmith, 2025-05-23:
It seems to be resolved. Now, if I press the "record" button immediately after recording a previous video, there is a long delay before the "ding" sound plays and it actually starts recording.

I'm assume that it is refusing to record a new video until it's finished processing the old one now, rather than just recording and dropping frames.

MHShetty, 2025-05-23:
Hi @cullumsmith 

Glad to know the final video no longer has any jerkiness at the beginning, but I'm really unable to reproduce either of the behavior on my end.

Please do let us know if the long delay issue still continues to persist.

Thanks a lot for trying out the camera app and reporting this issue!

closed 2025-05-29 by cullumsmith