GrapheneOS/Camera #492: Show (possible) QR code scam warning before scanning codes

robovoice1, 2024-12-14:
QR code scams are on the rise (in the bus, train, station, restaurant, parking ticket machines...)

This warning (feature) will not prevent scams, but keep the awareness of such possible scams high.
Usually if there are additional stickers above the original ad/poster something is fishy.
Qr codes with Included URLs starting with http:// instead of https:// are also an indicator that something is wrong.

MHShetty, 2025-05-16:
Hi @robovoice1,

We could start with detecting unusual characters to confirm and make it harder for the user to just click on any link.

Any further suggestions/cases that could have a solution for this issue and isn't hard to maintain in the long run could be considered while implementing this feature
 
Thanks for trying out our camera app and filing this issue!

MHShetty, 2025-05-16:
Hi @robovoice1,

We could start with detecting unusual characters to warn the user about it for actions such as opening a URL on the browser and make it harder for the user to just click on any link accidentally.

Any further suggestions/cases that could have a solution for this issue and isn't hard to maintain in the long run could be considered while implementing this feature
 
Thanks for trying out our camera app and filing this issue!