GrapheneOS/Camera #501: Camera error message

0l1e, 2025-01-09:
Hello. I often get this error when using the camera. Hope its possible to fix. It works to close the app and restart it.
[image]

MHShetty, 2025-02-02:
Hi @0l1e,

When exactly do you encounter this issue?

Thanks for using our camera app and reporting this issue!

0l1e, 2025-02-02:
Hi.
Well it happens after taking several photos, this error message pops up. I close the app and restart it and it works again.
Didn't have this issue with the original camera app.

Regards Olle

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

MHShetty, 2025-02-03:
Hi @0l1e,

Sorry to hear that you are facing this issue. The source of this issue still seems to be a bit uncertain. 

Do you repeatedly have to tap the capture button for this issue to occur or does the error show up even if the capture requests are not so continuous/frequently repeated in nature?

MHShetty, 2025-05-16:
Hi @0l1e,

Are you still facing this issue?