GrapheneOS/Camera #513: Ability to record slow-motion

Fireeyeeian, 2025-02-23:
I use slow motion to diagnose machine timing issues a lot and the ability to record slow-motion would be a greatly appreciated feature. 

The pixel camera seems to allow this but this doesn't seem to be implemented yet.

Thovthe, 2025-04-14:
I think this would be covered by Issue #51. I'm sure many people are looking for this functionality.  This issue could be closed in favor of the earlier one.

MHShetty, 2025-04-14:
Hi @Thovthe,

Yes we do have supporting slow motion videos in our road map. However simply setting the fps to a lower one won't perfectly help in creating slow motion videos. Will update this issue once the feature gets implemented from our end.

Thanks for trying out our camera app!