GrapheneOS/Camera #516: Secure Camera on Pixel OS crashes when selecting Night

FID02, 2025-03-07:
When the Night option is selected, Secure Camera on Pixel 8 is closed, with the OS reporting that the app was closed. Pixel OS doesn't give a crash log, so I generated a bug report with Developer options and will send it to matchbox on Matrix.

I have turned the device off, waited one minute, turned it on and tried again, issue remains.

Night Sight works fine with Pixel Camera.

*Version info*
app.grapheneos.camera.play version 81
Pixel 8
Pixel OS BP1A.250305.019

*Steps to reproduce*
1. Open the app
2. Select "NIGHT"

MHShetty, 2025-03-07:
Hi @FID02,

Could you please provide the exact/full stack trace of the logs related to the camera app closing?

Thanks a lot for trying out our camera app and filing this issue!

FID02, 2025-03-07:
@MHShetty How do I do that on stock Pixel OS? I couldn't find a tombstone in the bugreport zip that I sent to matchbox, even though I generated it after having reproduced the issue several times.

FID02, 2025-03-07:
Ok, it actually happens on GrapheneOS 2025030500 as well, on a Pixel 6a.

[Error in Camera dc824f39ea03.txt](https://github.com/user-attachments/files/19125630/Error.in.Camera.dc824f39ea03.txt)
[Camera log 3d06c592d1ca.txt](https://github.com/user-attachments/files/19125631/Camera.log.3d06c592d1ca.txt)

MHShetty, 2025-03-07:
Hi @FID02,

Thanks for sharing the concise logs regarding the crash.

We have filed a bug report to the CameraX team regarding the same here: https://issuetracker.google.com/issues/401460276 

Once the issue gets fixed at the level of the library, it should get resolved as we upgrade to the next version of the library.

For the time being, we have downgraded to a lower version of the library that should fix the issue in the next release.

Thanks a lot for your valuable time and help!