GrapheneOS/Camera #522: Uncroopped mode with lens selection instead of continuous zoom

dissociativity, 2025-03-18:
I would strongly prefer to be able to select my lenses and then manually crop later instead of having this awful continuous zoom where it's difficult to reset to uncropped ultra wide, main and zoom cameras.

This would make the camera app substantially more usable, as it stands I have to avoid using it due to this issue

dissociativity, 2025-03-19:
A return to 1x at the very least given how much better the main sensor is than the ultrawide would be very beneficial, as a photographer I'd like to be able to take raw format photos with the uncropped sensor of possible however

MHShetty, 2025-05-16:
Hi @dissociativity,

Hope you're doing well. Just a bump to this issue. Please let us know if the issue implies something apart from what's described here above

Thanks a lot for your time and support!