GrapheneOS/Camera #529: Where is the ability to change file formats?

NastyFlytrap, 2025-03-30:
I want to use png not jpeg. I really dont think thats a big ask

NastyFlytrap, 2025-03-30:
> We plan to add support for other formats but PNG is very bad at handling photos and there are better lossless image formats for this purpose such as the lossless variant of WebP.

Can you elaborate on why png is so bad?

Or go into a breakdown on the pros and cons of the formats worth talking about? Im interested to hear your thoughts about them

NXTler, 2025-10-15:
Not sure if it's relevant, but QOI is also an incredible lossless format with very fast encoding and decoding, while being very simple.
https://qoiformat.org/

NastyFlytrap, 2025-10-16:
> Ah, I wasn't aware that AV1 (and therefore parts of AVIF) is already the dominant format for cameras. Yeah, the compression is a real bummer here. I just wanted to share this interesting format to bring something new to the mix.

I mean, it IS interesting, but as he said, if the upside is that it uses little resources, for only being marginally smaller than png, then its not going to see any adoption. Besides, modern phones dont need to save on the photo compression hardware, they are plenty strong as is for something like that. Maybe for like an arduino or something...