GrapheneOS/Camera #544: Camera recording gets cancelled by power save mode

NicoDietrich, 2025-05-11:
**Problem**: If power save mode is enabled or disabled during a video recording, the app cancels the recording and goes back to the CAMERA tab. The video were saved in my last tests, but I am not sure if this is always the case.

**Expected behavior**: The recording should continue.

**Steps to reproduce**:

- Start a recording
- enable/disable power save mode (manually or automatically)

Current OS build: 2025050700
Current camera version: 83

MHShetty, 2025-05-11:
Hi @NicoDietrich,

Which device are you currently using to reproduce this issue? 

I'm unable to reproduce this on my Pixel 7A unfortunately

It could be possible that the OS puts the activity in background/for re-creation when power saving mode is enabled/disabled causing the video recording to stop since we currently don't support recording videos in background for the time being

Thanks for trying out our camera app and reporting this issue!

NicoDietrich, 2025-05-12:
Hey,
I use a Pixel 6a. I will edit the top post and add this. Sorry!

I tested this with my partner's Pixel 6a, also running the Camera with Version 83 but with GrapheneOS build 2025050300, and was surprised that I also could not replicate it. However, I am quite sure that the GOS version is not the issue because I noticed the same bug ~6 months ago.

I tried to diff the logs but couldn't find anything.

[Camera log 1d1ebbd4594a_working.txt](https://github.com/user-attachments/files/20173974/Camera.log.1d1ebbd4594a_working.txt)
[Camera log 5567a38e63fa.txt](https://github.com/user-attachments/files/20173975/Camera.log.5567a38e63fa.txt)

MHShetty, 2025-05-15:
Hi @NicoDietrich,

Really unsure what's causing the issue. I have tried reproducing this on my 7A and 4A but unable to reproduce this issue on either devices unfortunately

NicoDietrich, 2025-05-16:
Hey @MHShetty,
thank you for your testing!

As I mentioned in the previous post, I couldn't even replicate it on another 6a.
It seems like this is somehow related to specifics of my setup (which is nothing special to be honest :D).
I'll try to compare the two phones to see if I can find anything.

MHShetty, 2025-05-23:
Hi @NicoDietrich 

Ah no worries at all, please do let us know if you're able to figure out a more consistent way of reproducing this issue. Although I feel it might be related to the app currently not supporting recording in background but really unsure since I'm unable to reproduce the issue on my end.

Thanks for trying out our camera app and reporting this issue!

MHShetty, 2026-01-02:
Hi @NicoDietrich,

I'm closing this issue for the time being given it could have been a misunderstanding (considering the inactivity). Please do let us know if that's not the case otherwise.

Thanks a lot for trying out our camera app and reporting this issue!

closed 2026-01-02 by MHShetty