GrapheneOS/Camera #555: Add option

Men11312, 2025-07-06:
1. Is it planned to enter a watermark in the form of a date .... 06.25.2025.
2. Is it possible to be a jobly -generated value of brightness during adjustment ..
For example, when we reduce the brightness, it would show the figure with a minus sign (-1 ...- 2 ..., etc.). 
When we add brightness ... with a plus sign (+1 ...+2.. and etc.)
And then it is not clear what is the brightness.
3. It is advisable to make a lock for brightness (while the camera is turned on, the brightness should not change)

MHShetty, 2025-07-06:
Hi @Men11312,

I think you have filed multiple different issues within the same issue which kind of makes it hard to track progress for different issues and for someone new to know whether a given issue/feature request has been previously filed or not.

1. No, supporting watermarks on captured images is currently not in our plan. Although please feel free to file a separate feature request for the same. If multiple people have a valid use case or need it, we could consider including it in our plan.

2. From what I am able to understand, you wish to have a way to know the exact value of exposure on the UI. We could please draw It in place of the + icon at the top of the seek bar or something similar.

3. We have raised a feature request to have official CameraX support for exposure lock (https://issuetracker.google.com/issues/208583141). Also there's already a feature request on our issue tracker for the same #71

I'll try making a PR soon for the feature mentioned in the second point soon.

Thanks a lot for trying out our camera app and reporting filing this issue!

Men11312, 2025-07-06:
Thank you very much for the answer !!!
I will very much wait for the exposure blocking option for a while when the camera is active (even if it is folded).