GrapheneOS/Camera #561: Gyroscope indicator rotates backwards in certain orientations

jpsalis, 2025-07-29:
Device: Pixel 8 Pro
Build number: 2025071900

When taking a photo with the camera angled to the left or upside-down, the onscreen gyroscope indicator tilts the wrong direction. It's also very slow to update compared to the stock pixel camera.

MHShetty, 2025-07-30:
Hi @jpsalis,

Could you please describe the "rotates backwards" behavior a bit in detail?

I have tried testing our camera app and the Google Camera app and have observed that the gyroscope orientation in horizontal orientation seems to incline opposite of the device's tilt whereas our camera app goes with the tilt of the device. The slow updating issue is being dealt with in a different branch and will soon make a PR for the same.

Thanks a lot for taking time to report this issue!

jpsalis, 2025-07-30:
Correct, the indicator should tilt opposite the calculated z-axis orientation of the device to indicate the horizon.

However with the device tilted at or near 90° (Camera on the left-side) or 180° (Camera on the bottom), the indicator tilts with the device, rather than against.

MHShetty, 2025-08-05:
Hi @jpsalis,

Could you please test the camera app for the gyroscope tilt direction related issue by downloading and trying the apk below:

https://drive.proton.me/urls/6RA4HFJHT4#LKhVl196d11Y

Thanks a lot for your valuable time and help!

MHShetty, 2025-08-10:
Hi @jpsalis,

The fix for this issue has been merged to the main. Please do let us know if you're still facing the gyroscope rotation direction related issues in the latest version of our camera app.

Thanks for your valuable time and help!

closed 2025-08-10 by MHShetty