GrapheneOS/Camera #563: Portrait and Face retouch error on Samsung s25

animiou, 2025-07-30:
Hi when I try to use Portrait or Face retouch I get an error everytime. 

osVersion: samsung/pa1qxeea/pa1q:15/AP3A.240905.015.A2/S931BXXS5AYF4_OXM5AYF4:user/release-keys
package: app.grapheneos.camera:83

x.T: Capture request failed with reason ERROR
	at q.M.c(Unknown Source:28)
	at k.x.p(Unknown Source:27)
	at androidx.camera.extensions.internal.sessionprocessor.AdvancedSessionProcessor$SessionProcessorImplCaptureCallbackAdapter.onCaptureFailed(Unknown Source:2)
	at com.samsung.android.camerax.advanced.extender.RequestCallback$1.onCaptureFailed(RequestCallback.java:87)
	at com.samsung.android.camerax.advanced.extender.BurstRequestCallback.notifyCaptureFailed(BurstRequestCallback.java:120)
	at com.samsung.android.camerax.advanced.extender.BurstRequestCallback.onCaptureFailed(BurstRequestCallback.java:242)
	at androidx.camera.extensions.internal.sessionprocessor.AdvancedSessionProcessor$CallbackAdapter.onCaptureFailed(Unknown Source:19)
	at q.X.onCaptureFailed(Unknown Source:9)
	at q.c0.onCaptureFailed(Unknown Source:20)
	at android.hardware.camera2.impl.CameraCaptureSessionImpl$1.lambda$onCaptureFailed$5(CameraCaptureSessionImpl.java:722)
	at android.hardware.camera2.impl.CameraCaptureSessionImpl$1.$r8$lambda$nSI43m0hRmskZUsAoDo3H_ITffg(Unknown Source:0)
	at android.hardware.camera2.impl.CameraCaptureSessionImpl$1$$ExternalSyntheticLambda3.run(D8$$SyntheticClass:0)
	at C.f.run(Unknown Source:481)
	at C.f.a(Unknown Source:81)
	at C.f.run(Unknown Source:453)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
	at java.lang.Thread.run(Thread.java:1012)

animiou, 2025-07-31:
> Hi [@animiou](https://github.com/animiou),
> 
> This seems to be quite an unexpected issue. We have reported this further to the CameraX team.
> 
> Further detailed status of this issue can be tracked on https://issuetracker.google.com/issues/435220188
> 
> Any major updates for the same will be posted here
> 
> Thanks a lot for trying out our camera app and reporting this issue!

animiou, 2025-08-04:
> Hi [@animiou](https://github.com/animiou),
> 
> Could you please share your exact model number of your device?

SM-S931B/DS

MHShetty, 2025-08-05:
Hi @animiou,

Could you please provide the bug report while this issue occurs using `adb bugreport`? Also a recording while switching to the problematic mode attempting to take a picture multiple times could help in confirming that this issue always occurs with 100% reproducibility

The CameraX team could help us forward and communicate this issue to the Samsung Team for which they need the above details

Sorry for the inconvenience and trouble, and thanks a lot for your valuable time and help!

MHShetty, 2025-08-05:
They're unable to reproduce the issue on their end and we don't have a device to reproduce this issue unfortunately which is making resolving this issue kind of harder unfortunately (https://issuetracker.google.com/issues/435220188#comment7)

animiou, 2025-08-05:
> Hi [@animiou](https://github.com/animiou),
> 
> The CameraX team has shared a debug apk to test for this issue. If you're still facing this issue it's most likely an OEM related issue. Please do let us know if you're facing the same issue with apk present in the given link
> 
> https://issuetracker.google.com/issues/435220188#comment9

Thank you the apk solved the issue